sfuphantom/hardware #6: Prototype Latching Circuit

gabesoares, 2018-02-04:
Order components and prototype JK flip flop latching circuit
[image]

gabesoares, 2018-02-04:
Desired operation: fault signal is sending 0 as all is good, then a fault occurs and the signal sends a 1 (+5V) for a second and then drops back to 0.. We want this 1 to stay latched forever... 

JK Flip Flop Logic: 
If J=0, K=0, then output Q (& Q bar) stays latched to whatever it was before
If J=1, K=0, then output Q is set to 1, and Q bar is set to 0
If J=0, K=1, then output Q is set to 0, and Q bar is set to 1
If J=1, K=1, then output Q and Qbar are both toggled to the opposite of what they were before

NOTE: A clock signal is needed for the JK flip flop, a 555 timer can be used to generate the clock, but the exact circuit will need to be looked up

The push button attached to the K input will allow the output Q to be reset to 0, and Q bar to be reset to 1.
The red error LED should light up if the fault signal sends a 1 and the output is latched, and turn off when the push button (reset) attached to the K input is pressed. 

The green LED should be on in normal operation, and as soon as the fault signal sends a 1. It should stay off until the reset push button is pressed. (This green LED simulates the relay coils of the AIRs)

gabesoares, 2018-02-08:
Prototype this latching circuit as well, talk to Will to learn more about it:

[image]

Will-McKnight-604, 2018-02-14:
Updated Latching Circuit

[image]

gabesoares, 2018-02-14:
Latest latching circuit
[image]

gabesoares, 2018-02-14:
Will need high side driver instead of FET, since vgs will be tough to get without a charge pump, especially with a microcontroller output:

This one looks good, Rds(ON) a little high but not really significant for low current and turning on a relay, it also has a lower pin count and less external components than the other automotive TI ones
http://www.ti.com/product/tps1h000-q1

Will-McKnight-604, 2018-02-20:
Hanieh is taking the flip-flop circuit and prototyping that one, I'll be doing the latest latching circuit