sfuphantom/hardware #55: Question: How effective is the TSAL?

michaelruppe, 2019-08-07:
Hey friends,
I'm working on my own comparator-based TSAL, but your solution is attractive owing to it's simplicity. If you have the time spare, I have a couple of questions: 

- Does it trigger at a repeatable and accurate voltage? (both the turn-on and turn-off)
- Does the circuit need tuning? I imagine that for the small currents at the optocoupler LED, small manufacturing differences between optocouplers of the same type could change the trigger levels substantially. What have you experienced?
- How's the power dissipation in the input voltage-divider? 
- What voltage accumulator are you running?

I appreciate any time you take to answer these questions - I know well how short time is when building a race car.
If you prefer to answer outside of the public domain you can reach me at:
michael.ruppe@live.com.au

Thanks, and best of luck.

Khoideptrai123, 2019-08-14:
Hi Micheal,
My name is Khoi from Team Phantom FSAE.  We are still testing the TSAL right now, so the result is incomplete!
For your question, I have a couple of answers:
- The TSAL works in the range of 92V and above. We start testing the HV below 92V and the optocoupler does not work in accordance with its behaviors. However, our HV system runs from 120V to 168V so we have enough margin to maintain the behaviors of the optocoupler.
-For us, we use 168V for the voltage accumulator, so for the divider, we just pick the resistors that could maintain the power ( as P = V^2/R ) so the power dissipation is around 0.83W which is okay for our PCB.

I hope this answers your question. Please feel free to contact us if you have any questions.
Thank you and wish you the best, Micheal
 
Khoi,
Electrical member of Team Phantom

michaelruppe, 2019-09-03:
I appreciate the response, Khoi :thumbsup:
In the end I stuck with my [comparator-based design](https://github.com/michaelruppe/FSAE/tree/master/TSAL) to keep the switching threshold as close as possible to the 60V (Australian competition).